grblHAL/STM32F7xx #1: Have you not considered the H750VBT6 chip? It is 480Mhz, and it is cheap enough

hankecnc, 2021-06-07:
Have you not considered the H750VBT6 chip? It is 480Mhz, and it is cheap enough

hankecnc, 2021-06-07:
The iMXRT1062 is too expensive for me, and the BGA package is not conducive to soldering. I can't solder the PCB board at home, and the chip price is too high and it is not conducive to selling. If the price is too high, they may switch to buying a more mature MACH3 board.

hankecnc, 2021-06-07:
Is your version of grbl hal optimized for high-performance ARM chips? Because I know that grbl was first written on an 8-bit arduino, and it made many compromises due to the performance and memory of the microcontroller.

terjeio, 2021-06-07:
> Have you not considered the H750VBT6 chip?

No, but I checked out the specs of a H747 some time ago. You want to write a driver?

> Is your version of grbl hal optimized for high-performance ARM chips?

That depends on what you mean by optimized? Most of the drivers uses direct register access for GPIO and, at least for time critical parts, I try avoid overhead. It is also a reason for why I have not ported the code to C++

Other parts have been optimized to take atvandage of the 32-bit architecture as well, e.g. the planner and step buffers has been changed into linked lists instead of using indexed access. More memory used but faster.

>  it made many compromises due to the performance and memory of the microcontroller.

I believe most of these optimizations was to strip down functionality so that the code would fit available memory, not related to speed. grblHAL has added many standard G-codes such as canned cycles and the G59.x coordinate systems. More are planned. 

Perhaps the most important feature of grblHAL is that it has an architecture that allows expansion without altering the core, either at the driver level or as [plugins](https://github.com/grblHAL/plugins). E.g. a ftp daemon was recently added to networking to allow up/download of files to/from a SD card.

hankecnc, 2021-06-07:
> > 有没有考虑过H750VBT6芯片？
> 
> 不，但我前段时间查看了 H747 的规格。你想写一个驱动程序？
> 
> > 您的 grbl hal 版本是否针对高性能 ARM 芯片进行了优化？
> 
> 这取决于您所说的优化是什么意思？大多数驱动程序使用 GPIO 的直接寄存器访问，至少对于时间关键部分，我尽量避免开销。这也是为什么我没有将代码移植到 C++ 的原因
> 
> 其他部分也进行了优化，以利用 32 位架构，例如，规划器和步骤缓冲区已更改为链接列表，而不是使用索引访问。使用更多内存但速度更快。
> 
> > 由于微控制器的性能和内存，它做出了许多妥协。
> 
> 我相信这些优化中的大部分都是为了精简功能，以便代码适合可用内存，而不是与速度相关。grblHAL 添加了许多标准 G 代码，例如固定循环和 G59.x 坐标系。更多的正在计划中。
> 
> 也许 grblHAL 最重要的特性是它的架构允许在不改变核心的情况下进行扩展，无论是在驱动程序级别还是作为[插件](https://github.com/grblHAL/plugins)。例如，最近在网络中添加了 ftp 守护程序，以允许向/从 SD 卡上传/下载文件。

I do not have the ability to write a driver, I will only use IDE software to open your source code, click to compile, and then download to the microcontroller to run.
I used this firmware https://github.com/robomechs/6-AXIS-USBCNC-GRBL before, and it works fine, now I am trying your grblhal, thank you for your work for grbl

hankecnc, 2021-06-07:
I often see cnc control cards sold by merchants that indicate the output KHZ. I am confused about this. I don't know what that means. Does the maximum output KHZ capacity depend on the operating speed of the single-chip microcomputer?

terjeio, 2021-06-07:
It is the max. rate of the step pulses sent to the stepper driver. Yes, it is limited by the operation speed of the MCU - but also other factors such as interrupt latency and how the code is written. Max. rate achievable is also dependent on the gcode, if lots of small movements for laser engraving then buffer sizes plays into it as well.

Note that higher speed step rates requires hardware able to handle them, some drivers limits the pulse length to 2.5&micro;s for both on and off - limiting the max usable rate to 200 kHz.

Standard grbl limits min pulse with to 3&micro;S, grblHAL to 2&micro;s - I have seen claims that step rates of 500kHz or more is achievable with some 32 bit ports. This is simply not possible with grblHAL, and surely not any other unless the min. pulse with is reduced to 1&micro;s or less.

closed 2021-09-06 by hankecnc