grblHAL/STM32F7xx #7: Clock Source (HSE or HSI)

gxdeange, 2022-07-13:
Hi terjeio,

Can I please ask if the firmware will use the HSE or HSI as the clock source.

I assume that because this is a Nucleo board that it will use the HSI? 

Regards
George

terjeio, 2022-07-14:
It uses [HSE](https://github.com/grblHAL/STM32F7xx/blob/d14877250d83f84ddc5abb3bf3086de354d99969/Src/main.c#L45) sourced from ST-LINK. See section 6.8.1 in the Nucleo 144 user manual (UM1974).

gxdeange, 2022-07-14:
Can the firmware be configured for HSI (similar to how it is done for the
446 Nucleo)?

Regards George

terjeio, 2022-07-14:
Yes, IMO the best way is to create a new project to tweak the settings and then copy the clock setup code to the driver.

gxdeange, 2022-07-14:
Ok.

Would you be able to advise which driver file?

I can look at the HSI clock settings and adjust for maximum performance.

Cheers and have a great day
George

terjeio, 2022-07-14:
> Would you be able to advise which driver file?

[main.c](https://github.com/grblHAL/STM32F7xx/blob/d14877250d83f84ddc5abb3bf3086de354d99969/Src/main.c#L38-L84)

Have a great day too!

gxdeange, 2022-07-14:
Awesome.

Cheers George

gxdeange, 2022-07-16:
All Good

closed 2022-07-16 by gxdeange