grblHAL/STM32F7xx #8: Possible issue with .elf file

gxdeange, 2022-07-30:
Hi Terje Io,

I hope you are well.

I have attempted to flash a STM32F746 Nucleo board with the compiled .elf file but could not get it to upload.

I did however manage to get a previous compiled file (for the STM32F411) to succesfully upload to the Nucleo board (even though it is not the correct file).

I created a my_machine_map.h and ensured the BOARD_MY_MACHINE is not commented out but I suspect that it is not compiling properly or selecting my_machine_map.h file.

Regards and have a great day
George

terjeio, 2022-07-31:
Is it flashing via ST-Link from the debug tool in STM32CubeIDE that fails? 

BTW I've just checked the build parameters, the Release build is not properly set up - I'll fix that. For now use the Debug build.

Note that if you [add binary output](https://stackoverflow.com/questions/57017703/how-do-i-generate-a-binary-file-of-the-stm32-code) you can drop the .bin file on the _NODE_F756ZG_ drive in the file explorer to flash the image.

gxdeange, 2022-07-31:
I am flashing it via a ST-Link from the STM32Cube Programmer for which I have had great success with.

Cheers and have a great day
George

terjeio, 2022-07-31:
Fails in what way then?

Zip and attach your map file and I can try it - I am currently working with a F756 board (WebUI) so it is on my bench.

gxdeange, 2022-07-31:
When I try and flash the F756 binary. If I try a previous binary (such as the F446) on the STM32F746 nucleo board the flash oiperation works. I understand it is not the correct binary but i just wanted to verify it wasn't something I was doing wrong.

BTW I am using the STM32F746 which is effectively the same MCU expcept for the crypto.
Cheers G
[my_machine_map.h.zip](https://github.com/grblHAL/STM32F7xx/files/9228176/my_machine_map.h.zip)

terjeio, 2022-07-31:
`GPIO_BITBAND` has to be replaced with `GPIO_SINGLE` in _my_machine_map.h_ as the F7 does not support bitbanding.
Building and flashing succeded for me and I can connect to the board (via the USB connector at the same end as the ethernet jack).

gxdeange, 2022-07-31:
Cool. I have made the changes but I still cannot get the board flashed.

Do I need to download any new files from GitHub?

terjeio, 2022-07-31:
What exactly is going wrong? You get error messages when flashing?
Have you tried dropping the .bin on the drive presented by the board as I hinted above?

gxdeange, 2022-07-31:
Oops. My apologies. Added the option to covert to Binary and it looks like it has flashed succesfully.

Cheers and have a great day.
George

terjeio, 2022-07-31:
Ok, great.
A tip, add [BOARD_NAME](https://github.com/grblHAL/STM32F7xx/blob/f6922b9d1c3c23922520f61e78509953f4cb88a6/Inc/uno_map.h#L30) to your map file and it will show up in the `$I` output. You can use this to verify that your board map is actually compiled in.

gxdeange, 2022-07-31:
Awesome. Thanks

George

gxdeange, 2022-07-31:
I have just downloaded the latest files and attempted to compile and I am getting a bunch of errors (attached)
[image]

terjeio, 2022-07-31:
You have to [pull in the submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line). Did you do that?

gxdeange, 2022-07-31:
I have just downloaded the latest submodules / plugins and it is now throwing this error.

Cheers
George
[image]

terjeio, 2022-08-01:
> I have just downloaded the latest submodules / plugins and it is now throwing this error.

Oops, a regression introduced by me, should be ok in the latest commit. Sorry about that.

gxdeange, 2022-08-01:
Cool. Have updated everything and attempted to compile and it is throwing this error.

Regards George
[image]

terjeio, 2022-08-02:
This is odd, I have not modified the file containing main() - delete the project and import it anew as something has been corrupted? It compiles ok for me with the default settings and with WebUI enabled as I have done fair bit of testing of that lately.

gxdeange, 2022-08-02:
Ok. I will do that later tonight and report back.

Regards George

gxdeange, 2022-08-02:
OK. I have downloaded everything and it now compiles with just a few Warnings. I believe I know why it was failing now: I had updated the clock settings in main.c to use the HSI. When I changed it back to the original HSE config it compiled without errors.
Is it possible to add additional build options such as F746 / F756 Nucleo so that the HSI clock is used?

Cheers and thank-you for all your help
George
[image]

terjeio, 2022-10-14:
Sorry for late answer.

> Is it possible to add additional build options such as F746 / F756 Nucleo so that the HSI clock is used?

Yes, right click the project folder and select _Build Configurations > Manage_ to add new configuration(s) and then add symbol(s) to them for enabling appropriated code in `SystemClock_Config()` in _main.c_.

closed 2024-02-29 by gxdeange