grblHAL/STM32F7xx #13: hello terjeio, sending you an email was unsuccessful, so I'm here to discuss something with you

hanke-cnc, 2023-01-16:
Hello, hello terjeio, I am from China, I am a machining center operator, I like cnc very much, I started DIY cnc 3 years ago, it is very interesting,

Compared with CNC, I think CNC system is more interesting, so I started to learn to contact grbl, because it is open source

Later, I started to learn how to make circuit boards. I made circuit boards for ESP32 GRBL and STM32F4 GRBL_HAL and sold them. Although it did not bring much income, I found it very interesting.

I have made an embedded linux controller with a touch screen based on STM32 F4 GRBL_HAL and Candle, although I have not sold a few, but I have considered future development,

No one knows grbl better than you. I want to adopt a cooperative form. I don’t know if you can add some functions to grbl_hal. Is it difficult to implement? These functions do not need to be open source. I don’t need these functions yet. Now I’m just asking

Function 1: 5-axis RTCP is something that many Chinese buyers consult. This function is of great commercial value. I don’t know how difficult it is to realize? I only know that RTCP is a matter of kinematics.

There are other functions of industrial controllers, such as radius compensation, S acceleration and deceleration, G84 servo spindle tapping,

The above function is just to ask how difficult it is to implement terjeio? It is not particularly necessary, and the current version of grbl_hal is enough

terjeio, 2023-01-18:
First, post your email address in a comment and remove it afterwards so I can give it a try.

> 5-axis RTCP

This might be hard to implement since grblHAL does not have access to the complete gcode program (unless it is stored on a SD-card accessible by the controller). From a quick glance complete access (or at least a sensible amount of blocks ahead) is needed?

> radius compensation

Same as for RTCP above?

> S acceleration and deceleration

Possible and I may implement that at some future date if not someone beats me to it. Somehow related is that I want to decouple the planner and stepper code from the core so it can be run on a separate MCU (core) or a FPGA.

> G84 servo spindle tapping

I am looking into how to implement G33.1 which is harder to get right than G84. When done G84 should be easy. My main problem with developing this is that I do not currently own equiment to run tests on...

> The above function is just to ask how difficult it is to implement terjeio?

Impossible to tell before starting the work and getting a sense of how complicated it is and how much testing/debugging it reqiures. Testing/debugging can be far more time consuming than writing code...

hanke-cnc, 2023-01-18:
> First, post your email address in a comment and remove it afterwards so I can give it a try.
>
>
> This might be hard to implement since grblHAL does not have access to the complete gcode program [...]

Oh, don’t feel embarrassed, I’m just temporarily asking terjeio about the possibility of realizing them. I don’t need it now, because I have already made an embedded Linux CNC system based on your STM32_F4_grbl_hal and QT_Candle. If the sale goes well, I want to talk to you in the future Paid cooperation adds some functions to it, so that it can go further,

hanke-cnc, 2023-01-18:
Oh, don’t feel embarrassed, I’m just temporarily asking terjeio about the possibility of realizing them. I don’t need it now, because I have already made an embedded Linux CNC system based on your STM32_F4_grbl_hal and QT_Candle. If the sale goes well, I want to talk to you in the future Paid cooperation adds some functions to it, so that it can go further,

Do you have another email? For example, Google email, the previous email failed to send

closed 2023-01-18 by hanke-cnc