grblHAL/STM32F7xx #21: Duplicate definition causing compilation error using uno-map.h

a-m-shaw, 2025-01-10:
Hi,

I believe there's a duplicate block of code in `boards/uno_map.h` that's causing compilation failure. It looks like lines 150 to 186 may contain some copy/paste errors; I'm not immediately clear on what the intended purpose of `N_ABC_MOTORS` is but it doesn't look like it should control the coolant defintions.

FWIW: I'm trying to set up a Nucleo-F767ZI with an old gShield v5b, and I _think_ I need to use the "Generic Uno/Nucleo-144" board for that, but the web builder currently errors out with the below. I understand that this hardware driver is fairly experimental so I'd be happy to contribute my test results (and would appreciate advice if I'm doing anything wrong!)

Thanks.

```Command failed: pio run -etest -cjob.ini
In file included from Inc/driver.h:120:0,
from bluetooth/hc_05.c:24:
./boards/uno_map.h:182:0: warning: "COOLANT_MIST_PORT" redefined
#define COOLANT_MIST_PORT GPIOF // A4

./boards/uno_map.h:166:0: note: this is the location of the previous definition
#define COOLANT_MIST_PORT AUXOUTPUT5_PORT```

terjeio, 2025-01-10:
Thanks for reporting. Please report back regardless of how testing goes.

a-m-shaw, 2025-01-10:
Thanks for the quick response! This now looks good to me and I've proven basic jog control over three axes. I'll be testing more robustly in the coming days. 👍

closed 2025-01-10 by a-m-shaw